Lamarqe/ha_openems #1: Energy Sensors not available for Home Assistant Energy Dashboard

TheSerapher, 2025-02-22:
While migrating to your integration most things worked but I am unable to add the new sensors to the energy dashboard. HA expects the following energy class units:

```
Pick a sensor which measures grid consumption in either of cal, Gcal, GJ, GWh, J, kcal, kJ, kWh, Mcal, MJ, mWh, MWh, TWh, Wh.
```

Sensors are added to HA with `Wh_Σ` instead, which also means they are never marked as `Energy` sensors device classes: https://github.com/Lamarqe/ha_openems/blob/main/custom_components/openems/helpers.py#L9-L10

Maybe a simple transformation before this line from `Wh_Σ` to `Wh` would be enough? https://github.com/Lamarqe/ha_openems/blob/main/custom_components/openems/sensor.py#L91

TheSerapher, 2025-02-23:
I think there is one more additional step to do for all `energy` type device classes: They must be set to `state_class: total_increasing` in order for statistical functions like `change` to work on them. I noticed my graphs for energy related information were empty and testing with the statistics card in Home Assistant I noticed I could't select the `change` function. They behave like normal sensors, hence I assume the `state_class` is missing.

TheSerapher, 2025-02-28:
As mentioned, your solution supplied seems to have issues with all sensors going unavailable once your changes are added: https://github.com/Lamarqe/ha_openems/commit/16b1305dafa9ce1076750aef3415f6ae57a77a30 - I reverted back and things work as usual.

Without real knowledge I would think that not setting units explicitly everywhere, accessing them here returns None and it fails setting up the sensor? https://github.com/Lamarqe/ha_openems/blob/16b1305dafa9ce1076750aef3415f6ae57a77a30/custom_components/openems/sensor.py#L100

Wild guess, probably wrong. I am sure you find a solution :)

TheSerapher, 2025-02-28:
Noticed that your `sensor.py` also included a few changes related to disconnect handling which I also added (since I copy & pasted the file). I have a feeling the unavailable sensors happen due to the changes in the config flow and maybe missing variables? Probably best to reinstall the integration with the new version then?

TheSerapher, 2025-02-28:
Thanks for confirming, also reinstalled it and sensors work just fine. One thing I noticed though, since I completely removed and uninstalled the integration, reinstalling and re-enabling it didn't bring back my hidden sensors. I had enabled quite a lot of them (since they return useful information for my dashboards) I had to do the process again. No big deal since I don't think we will go through a full wipe often, just something to keep in mind when it does happen!

Sensors now show properly and can be selected in the Energy Dashboard, so the initial cause for this ticket seems resolved. Closing this for now :)

TheSerapher, 2025-02-28:
Spoke too soon, sorry.

Sum Values are not included in the unit checks hence they came back with the wrong unit (see screenshot). They probably need some special handling as well?

[image]

Lamarqe, 2025-02-28:
I have doubts that there is still a problem in the code.
Could you post an example entity detail from <youHASite>/developer-tools/state.

This is how it looks to me, and this is fully correct, I believe:

[image]

TheSerapher, 2025-02-28:
Maybe those were stray sensors from a previous installation since I wiped the integration. Not sure but the issue seems gone and they do report as Wh now. Not sure what happened here but I’ll close this 🙂

closed 2025-02-28 by TheSerapher