Lamarqe/ha_openems #7: Integration Logging

TheSerapher, 2025-02-28:
Looking through the code I noticed some print calls and was wondering if those shouldn’t be converted to a logger class by home assistant? I would also like to see more logging information in general to check what the integration is doing, like a failed connection and trying to re-establish.

Maybe some improvement for later 🙂

Lamarqe, 2025-02-28:
You are 100% correct. This integration does not fulfill quality standards. If you check the code, the structures are questionable and do all kind of hacks (eg private member access.).

That being said: Its not yet worth to track such a quality issue with a github issue. 
If things go as planned, this code will be repeatedly refactored. As part of that also these print call will be cleaned up.

Lamarqe, 2025-02-28:
Note detection of these issues is simple. The default Home Assistant devcontainer provides extensive linting and quality checks out-of-the-box.

Once its the time, I will "only" have to go through all the findings in the commit pre-checks.

closed 2025-02-28 by Lamarqe

TheSerapher, 2025-03-04:
Figured that's the case! Been running without issues so far but still have to test the reconnect feature. Waiting for a FEMS upgrade since this is a very common scenario for me where the connection is dropped, I usually don't restart the systems manually.