Lamarqe/ha_openems #8: Enum device class using Measurement state cass

TheSerapher, 2025-02-28:
Seeing these in my logs. Enum should be None but as it turns out any Energy unit should not be measurement and either total or total increasing. Makes sense I guess.

```
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:547
integration: Sensor (documentation, issues)
First occurred: 20:55:02 (19 occurrences)
Last logged: 20:55:02

Entity sensor.fems72404_evcs0_chargestate (<class 'custom_components.openems.sensor.OpenEMSSensorEntity'>) is using state class 'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise report it to the author of the 'openems' custom integration
Entity sensor.fems72404_evcs0_chargemode (<class 'custom_components.openems.sensor.OpenEMSSensorEntity'>) is using state class 'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise report it to the author of the 'openems' custom integration
Entity sensor.fems72404_evcs0_energysession (<class 'custom_components.openems.sensor.OpenEMSSensorEntity'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the author of the 'openems' custom integration
Entity sensor.fems72404_meter0_state (<class 'custom_components.openems.sensor.OpenEMSSensorEntity'>) is using state class 'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise report it to the author of the 'openems' custom integration
Entity sensor.fems72404_meter2_state (<class 'custom_components.openems.sensor.OpenEMSSensorEntity'>) is using state class 'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise report it to the author of the 'openems' custom integration
```

Lamarqe, 2025-02-28:
This should be fixed with v0.2.2. Does it still happen even using this version?

TheSerapher, 2025-03-01:
Updated, logs are clean besides the known issue with mHz not being supported 🙂 Thanks man!

closed 2025-03-01 by TheSerapher