Lamarqe/ha_openems #10: Reconnection Issue

TheSerapher, 2025-03-04:
Hey,

I just had a case where my FEMS sensors had no data, looking in the logs they didn't show anything related to connection issues - probably because no logging is in place to report a dropped connection?

The good news: I triggered a reload of the integration and the connection was restored. I do have the auto-reconnect version of the integration installed but can't supply more information.

Lamarqe, 2025-03-04:
I will add the losging to report when the connection gets lost and restored again.

With "no data" you mean all sensor were in status unavailable,  right? (That's the current implementation when the connection is lost)

TheSerapher, 2025-03-20:
Just ran the system upgrade to 2025.3.1 which triggered a restart and everything worked! Update installed without issues and the integration reconnected showing data again the minute FEMS was back. Well done!!

Not sure if we can close this already but it did work this time.

closed 2025-03-21 by Lamarqe