Lamarqe/ha_openems #11: Configuration via HA not working.

salomonde, 2025-03-21:
Hi,

unfortunately configuration is still not working after update to 0.4.0.

I attached the HA logs. Don't know, if this is any help. Let me know, how I can help to fix this issue.

Thanks in advance.

[home-assistant_2025-03-21T08-56-42.910Z.log](https://github.com/user-attachments/files/19385819/home-assistant_2025-03-21T08-56-42.910Z.log)

Lamarqe, 2025-03-21:
We need to find out which websocket address your OpenEMS system uses to communicate with the backend.
Then I can figure out it if this is a generic behavior, or if I need to provide an extra option which allows people like you to specify an individual value.

Please do the following:

1. open the browser console (ctrl+shift+I) 
2. open the network tab, select only WebSocket (WS) communication
3. open the OpenEMS WebUI
4. Select the websocket connection which was created
5. In the details view, headers field, you should see the websocket URL which is used. Please post the value here.

It should look similar to this: ws://192.168.x.x:1234/...

salomonde, 2025-03-24:
I followed the steps. It shows:

ws://192.168.178.43:8085/

When I login to the UI I get forwarded to: http://192.168.178.43/device/0/live

I can't find any file called "openems_config" on the server.

I am not quiet sure if I am using the correct login and password. There is no username for the Openems UI. I can just use "guest" or "admin" as login to login as guest oder admin. What should I use as username in HA?

Lamarqe, 2025-03-24:
The websocket URL looks good. That means that the correction done with v0.4.0 was definitely necessary and is correct.

Regarding the login: Please use username "x", and password "admin". This should work.
You definitely already had the correct login defined at some point.
As soon as you see a `KeyError: '0'` message, the login is correct (even though the initialization will fail).

> I can't find any file called "openems_config" on the server.
Yes, makes sense. I reviewed the code related to the `KeyError: '0'` message and can confirm that this is happening even before the config file gets written.

To analyze the bug which leads to the KeyError: Could I simply provide you with a python script that you run on one of your systems? Alternatively, I can also add log outputs directly to a new pre-release and you send me the home assistant logs when trying these.

salomonde, 2025-03-31:
Hi. Yes. Just send the python script over.

Lamarqe, 2025-04-04:
Can you please run the attached script (you can just run `python pyopenems.zip`) and send me the output of it?

[pyopenems.zip](https://github.com/user-attachments/files/19608286/pyopenems.zip)

Lamarqe, 2025-05-29:
Closing the issue as this is expected to be fixed in release 0.8.x and due to missing user feedback.

closed 2025-05-29 by Lamarqe