Lamarqe/ha_openems #15: Feature: Update Check

TheSerapher, 2025-04-16:
I was wondering if it’s possible to add the update notification to Home Assistant as well? Would be nice to be notified via HA that an update is available, I am fine with triggering the update through the FEMS app. Now I have to check there ever so often to ensure I am up to date.

Lamarqe, 2025-05-24:
This was fairly easy to implement: https://github.com/Lamarqe/ha_openems/compare/main...update_check

[image]

But it looks like [FEMS is currently broken](https://www.photovoltaikforum.com/thread/248064-ladeformulr-ist-unvollständig). So I don't want to update my system right now and test it.

As soon as I can properly test this feature, I will merge it.

TheSerapher, 2025-05-30:
Great work already! I wasn't even expecting to execute the update with Home Assistant, not to mention a progress bar at all. It's just a lot easier to see there is something available when using Home Assistant all the time than checking the FEMS app once in a blue moon. Thanks a lot for your work on this!

Lamarqe, 2025-06-06:
@TheSerapher : Did you already update to FEMS 2025.06.1?

Backround: After studying the backend behavior during this update, I have extended the implementation to cover also the progress part and the cleanup of the notification. This needs testing, but worst case I will need to wait another month until 2025.07.1 is out.

So I am looking for somebody who is will to try it and has an update pending.

TheSerapher, 2025-06-06:
> @TheSerapher : Did you already update to FEMS 2025.06.1?
>
> Backround: After studying the backend behavior during this update, I have extended the implementation to cover also the progress part and [...]

Already update, sorry mate :-( They might release smaller hotfix releases soon though, not uncommon to see .2 and .3 releases.

Lamarqe, 2025-06-07:
If you want to give it a try with an upcoming release: There is now a v1.0.0 pre-release available which contains the update feature.

TheSerapher, 2025-06-11:
I will install it now, there is an update available!

TheSerapher, 2025-06-11:
[image]

Clicked update, saw the update also starting in the portal itself. Progress reported 25% in the portal and in Home Assistant. It then restarted and became unavailable as expected, after booting up the system was back.

Lamarqe, 2025-06-11:
Thanks for the positive feedback. With that I will close this issue.