Lamarqe/ha_openems #17: Input Boolean instead of sensor for Boolean

TheSerapher, 2025-05-23:
Hey,

Just noticed that the

sensor.fems<id>_ctrlesssurplusfeedtogrid0_surplusfeedtogridislimited
[image]

is a sensor and not a Boolean instead Home Assistant.

I wonder if this could be changed since it would allow cleaner automation triggers 🙂

I think this was a special treatment for this entity before you added Ctrl entities?

Lamarqe, 2025-05-24:
Input boolean aims at controlling a state. However this channel looks like a regular read only states to me.

I guess you mean to change the plain sensor into a binary sensor, right? 
( https://www.home-assistant.io/integrations/binary_sensor/)

Lamarqe, 2025-05-29:
Implemented with https://github.com/Lamarqe/ha_openems/commit/49d6c3c6d065752da65f108fdb05f3c60ef03ace and https://github.com/Lamarqe/ha_openems/commit/4f16f354da23be2184389b1592bac186d21c7d75.

Will be part of release 1.0.x

closed 2025-05-29 by Lamarqe