Lamarqe/ha_openems #19: Potential Reconnect Issue

TheSerapher, 2025-06-03:
Hey,

I am having network issues on my Raspberry Pi 5 running HAOS which causes the network adapter to sometimes not forward traffic - I was debugging this for a while but can't really find a root cause. In one of those events, all integrations disconnected (e.g. Bosch, Unifi, OpenEMS) but only OpenEMS stayed offline. I had to restart home assistant to recover. Going through the logs I saw no messages from OpenEMS related to connection issues, which was strange. Is there any INFO output available from the integration during disconnect events? Would have been interesting to see why it failed to reconnect.

Maybe I am missing something but DEBUG shouldn't be the only option here since it looks like an ERROR worthy event that should be logged.

Running the latest version as of today.

Lamarqe, 2025-06-03:
Is this behavior reproducible when you pull the network cable from your RPI?

The integrations TCP heartbeat to the backend in is set to 5 seconds. So within 5 seconds after removing the cable, the connection loss should be detected and restored accordingly afterwards. When you enable debug logging, you should see according debug messages.

Note: I agree that these should not just be debug messages. I will change them to info messages.

TheSerapher, 2025-06-03:
Maybe ERROR messages even - since it does look like an error from the integration perspective if the connection drops.

They are very weird network errors. It's not a connectivity loss, the adapter never mentions a loss of connection, neither do I see anything in the Unifi logs that a device was disconnected. It's more like an I/O collision someplace in the Raspberry itself that causes short hiccups. Most apps recover, and OpenEMS usually too, but this one time it failed to reconnect. I tried to reproduce it but wasn't able to, hence I am kindly asking for a little more logging output so I have something to check the next time it happens: Connection lost, retrying message(s), potential error messages if the reconnect fails, anything else that might be worthy to log.

Thanks for looking into this! With better logs at hand I could give more help 😃

TheSerapher, 2025-06-03:
I've seen those too - but always just stamped them as bad connection handling in inactive browser tabs. It was an assumption since the app itself always triggers a full reload when switching between apps, even if it's just for a second. But it might be related and could be the same cause for the issues I saw.

I do see full connection loss to other smart home controllers, I would assume that the integration also sees a full connection loss.

Lamarqe, 2025-06-04:
> Maybe ERROR messages even - since it does look like an error from the integration perspective if the connection drops.

https://developers.home-assistant.io/docs/core/integration-quality-scale/rules/log-when-unavailable

😉

Lamarqe, 2025-06-07:
v1.0.0 contains log and debug messages which allow to trace the the connection status.

If you have issues with the connection stability, I recommend to switch the [connection type](https://github.com/Lamarqe/ha_openems?tab=readme-ov-file#main-configuration-dialog) to FEMS local monitoring. 

Background: I am able to reproduce a FEMS bug which entirely stops communication via the direct edge websocket connection on port 8085 until the FEMS websocket server gets restarted. 
I cannot reproduce this bug in FEMS/OpenEMS when using the regular local monitoring connection (port 80). 
I can expose this bug via another bug (https://github.com/OpenEMS/openems/issues/3169). Maybe this problem can also occur when the connection is instable and frequently gets reconnected due to your Raspberry PI network problem.

 _Side Note: I expect your networking issues to be caused by improper power supply. If I were you, I would first invest into a new power supply, in order to definitely rule out this possible root cause._

closed 2025-06-11 by Lamarqe