Lamarqe/ha_openems #23: Updated FEMS to 2025.7.4 and integration no longer works

dokschneider73, 2025-07-30:
I’ve just updated my FEMS system to Version 2025.7.4 and the integration stopped working with Error 500 Server got itself in trouble.
Made a reinstallation but no good. Any idea what might be the issue?

BR
Roland

Here is the log:

Logger: homeassistant.config_entries
Quelle: config_entries.py:749
Erstmals aufgetreten: 19:35:08 (4 Vorkommnisse)
Zuletzt protokolliert: 22:22:53

Error setting up entry 192.168.178.72 for openems
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/openems/__init__.py", line 148, in async_setup_entry
    await backend.prepare_entities()
  File "/config/custom_components/openems/openems.py", line 878, in prepare_entities
    await self.the_edge.prepare_entities()
  File "/config/custom_components/openems/openems.py", line 641, in prepare_entities
    await component.init_channels(contents["channels"])
  File "/config/custom_components/openems/openems.py", line 423, in init_channels
    await prop.init_limits(limit_def)
  File "/config/custom_components/openems/openems.py", line 305, in init_limits
    lower_limit = await self._compute_expression(limit_def["lower"])
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/openems/openems.py", line 329, in _compute_expression
    return float(Template("{{" + value_expr + "}}").render())
ValueError: could not convert string to float: 'None'

Lamarqe, 2025-08-02:
Can you please switch to v1.0.3, enable debug log and try to re-add your config entry afterwards (should still fail, there is no logic change yet, only extended tracing).
Afterwards, stop the debug log, open the log file and search for the string "_get_ref_value".

Then, please post the respective lines here.

Lamarqe, 2025-08-04:
Interesting. The Null response is indeed unexpected and the reason for the problem. Either there is a fix for this behavior or I need to implement a workaround solution.

Regarding EV charging vendor I meant the vendor of the charger. Keba? Hardy Barth? Something else?

The integration uses the value of `MinimumHardwarePower` for setting the lower limit value of the `ForceChargeMinPower` property. What is the value in the FEMS UI for you?

This is how it looks to me:

[image]

I wonder how this is calculated by the FEMS WebUI. Maybe I have to hard-code a default fallback value.

Lamarqe, 2025-08-04:
Okay, good that the problem is solved.

Still, its a poor behavior of the integration that it entirely stops working due to this small problem.
I will implement a fallback, same as you describe it from FEMS: Use 0 in case the backend does not provide a proper value.

Lamarqe, 2025-08-05:
Fixed withhttps://github.com/Lamarqe/ha_openems/commit/60b91e56db5545601e2e34403a1cc488ce5fcb1e.

closed 2025-08-05 by Lamarqe