Lamarqe/ha_openems #26: Entities for different strings missing?

HorizonKane, 2025-09-03:
Hi,

In the Fenecon App, I can see the power every string generates -  it I can‘t find corresponding entities. Am I missing something or are they not available?

Kind regards
Thomas

HorizonKane, 2025-09-04:
> I would suspect you are missing something. Nothing I can see from your report being wrong in the integration.

So…are their entities for the production of each string? What’s their name?

Lamarqe, 2025-09-04:
I still don't see the issue to the integration here.

Isn't this a regular support request to be discussed in a regular forum like openems.io or home-assistant.io?

Skeletitor, 2025-09-05:
@HorizonKane 
Do you not see this in your devices and services list in Homeassistant??

[image]

and 

[image]

HorizonKane, 2025-09-05:
@skeletitor Hi, thank you for showing. No I actually only see the first one, meta and sum, but no devices for charger 0 and charger 1.